Donkie/SpoolmanDB #5: RFC: Handling different spool weights for the same filament.

StuSerious, 2024-05-14:
Hey there Donkie!
It's refreshing to finally see some real effort being put into a centralized database, I have been waiting for this for a long, long time.

I manage a 3D printing research lab and have access to and can contribute hundreds of spools from tens of manufacturers.
While working on my first pull request to add Formfutura filaments, I remembered that they sometimes sell different spools for the same filament and weight product.
As also seen in [their technical spool information](https://formfutura.sharepoint.com/sites/downloads/Shared%20Documents/Forms/AllItems.aspx?ga=1&id=%2Fsites%2Fdownloads%2FShared%20Documents%2FDownloads%20Server%2FMaterials%2FFilaments%2FTechnical%20Spool%20Information&viewid=2c9648b9%2D9666%2D45d4%2Da59a%2D32077bd10762), they offer cardboard and plastic variants for the same product, each with different spool weights.

For example, the cardboard 2300g spool weighs ~[265g](https://formfutura.sharepoint.com/sites/downloads/Shared%20Documents/Forms/AllItems.aspx?ga=1&id=%2Fsites%2Fdownloads%2FShared%20Documents%2FDownloads%20Server%2FMaterials%2FFilaments%2FTechnical%20Spool%20Information%2FCardboard%20Spools%2Fff%2DSpool%5FSpecifications%2D2300g%5FCardboard%2Epdf&viewid=2c9648b9%2D9666%2D45d4%2Da59a%2D32077bd10762&parent=%2Fsites%2Fdownloads%2FShared%20Documents%2FDownloads%20Server%2FMaterials%2FFilaments%2FTechnical%20Spool%20Information%2FCardboard%20Spools), while the plastic 2300g one weighs ~[600g](https://formfutura.sharepoint.com/sites/downloads/Shared%20Documents/Forms/AllItems.aspx?ga=1&id=%2Fsites%2Fdownloads%2FShared%20Documents%2FDownloads%20Server%2FMaterials%2FFilaments%2FTechnical%20Spool%20Information%2FPlastic%20Spools%2Fff%2DSpool%5FSpecifications%2D2300g%5FPlastic%2Epdf&viewid=2c9648b9%2D9666%2D45d4%2Da59a%2D32077bd10762&parent=%2Fsites%2Fdownloads%2FShared%20Documents%2FDownloads%20Server%2FMaterials%2FFilaments%2FTechnical%20Spool%20Information%2FPlastic%20Spools).

I would like to hear your opinion on how to best handle such cases, as I've seen a steady rise in manufacturers providing cardboard spools as an eco-friendly option.

Thank you so much for all the work you're putting into all this, I look forward to contributing to the best of my ability!

Donkie, 2024-05-15:
I have now added an additional field to the weights object where you can specify the type of spool. So in your example you would do:
```json
"weights": [
    {
        "weight": 2300.0,
        "spool_weight": 600,
        "spool_type": "plastic"
    },
    {
        "weight": 2300.0,
        "spool_weight": 265,
        "spool_type": "cardboard"
    }
],
```

StuSerious, 2024-05-16:
Thank you so much! this seems perfect.

I'll push an initial PR with the filaments in question so you can take a look!

closed 2024-05-19 by Donkie