Donkie/SpoolmanDB #13: PSA: list of known spool & filament databases

StuSerious, 2024-05-16:
_WIP!_

- [Printables Database](https://www.printables.com/model/464663-empty-spool-weight-catalog) by [@Scuk_253012](https://www.printables.com/@Scuk_253012)
- [Onlyspoolz](https://www.onlyspoolz.com/portfolio/) hosted on https://github.com/jtrmal/spoolz by @jtrmal
- [PrinterMaterials Empty Spool Weight Database](https://printermaterials.com/empty-spool-weight-database/) by [John Arist](https://printermaterials.com/author/john/)
- https://spoolweight.github.io/

runtpacket, 2024-12-05:
https://www.printables.com/model/464663-empty-spool-weight-catalog/comments
https://3dfilamentprofiles.com/

ex-nerd, 2024-12-14:
Don't forget https://filamentcolors.xyz/

ezraclintoc, 2025-12-03:
Yo, could I try to make something to automatically add filaments from these sources (I am thinking particularly about 3dfilamentprofiles.com)