Donkie/SpoolmanDB #19: Potential Database and Schema Updates - Filled filaments, temperature ranges, material field, alloy codes.

YanceyA, 2024-05-22:
In the work to create entries for 3DXTECH filaments i ran across a few bits which could be addressed. I fully understand that making the database too complex is not desirable or actually beneficial to end users. 

-3DXTECH use temperature ranges for extruder and bed temp. The JSON schema only allows for a single integer. It would be nice to be able to input two values indicating a temperature range recommended by the manufacturer. Alternatively we could have a min and max values for extruder/bed (less preferred I think as that creates two new fields and begs the questions of what to enter when only a single value is given by the manufacturer). 

This probably doesn't matter as users set their own print temps through the slicer config and filament calibrations. For 3DXTECH filaments I just averaged the two band values. 

-I think we need a solution to deal with filled filaments. My initial suggestion is simply a "fill" parameter that is "null, glass fiber, carbon, fiber". The material name could then be  NAME {color_name} {fill}.  Most of the 3DXTECH entries have fill material and that is important to call out and it feels a bit clunky to only have it as text "Carbon Fiber" in the name.

Also some guidance on how to deal with the material field text with regards to fill and alloys would be helpful:

- For two filaments ABS and Carbon Fiber ABS should they both have material as ABS or should they be ABS and ABS-CF. My suggestion is to use the later as they are two difference materials and conforms to ISO 1043 marking standard.

- In general polymer alloys should be designated as PC+ABS and not PC/ABS to conform to ISO 1043 marking standards. So a blend with fill would be PC+ABS-CF or PC+ABS-GF. This gives us a basis of why to do it this way.

Donkie, 2024-05-23:
I could add new fields called e.g. "extruder_temp_range" and then you can either specify a single value or use this new field to specify a range. That should solve it for all use cases.

Yes, a fill field is a good idea. You can submit a PR to add that to the schema if you want. I dont think you need to add {fill} to the filament name though since that should be evident from the material name, where I agree we should go with as you said e.g. ABS-CF.

I also agree with your blend marking there. Good to keep it consistent.

StuSerious, 2024-05-23:
I've drafted an idea in https://github.com/Donkie/SpoolmanDB/pull/21

Donkie, 2024-05-28:
I think this is fully implemented now by Stu's PR

closed 2024-05-28 by Donkie