Donkie/SpoolmanDB #26: RFC: handling of refillable spools

StuSerious, 2024-05-27:
A lot of manufacturers like [Formfutura](https://formfutura.com/product/refill-pla/), [BambuLab](https://eu.store.bambulab.com/en-it/products/bambu-reusable-spool), [MatterHackers](https://www.matterhackers.com/store/c/mh-build-series-pla-refill-filament), [Prusament](https://www.prusa3d.com/en/category/prusament/?filters=%7B%22parameters%22%3A%5B%7B%22parameter%22%3A%22b027cf78-4d8d-4f63-9943-a6e4f6d7714e%22%2C%22baseColors%22%3A%5B%5D%2C%22values%22%3A%5B%2292546b3b-677e-4b0d-a4a1-afccf2fd6a3c%22%5D%7D%5D%7D) and more, have started offering reusable material in the last few years.

The idea is that you only buy a spool once, and then use refill "cartridge" filament in that same spool over and over again.
This also means that users may have multiple, different materials but all with the same spool weight.

Do you think there should be any particular way or addition to allow for these types of spools?
for example, a "reusable spool" parameter for manufacturers that provide it so that any UI or client can implement the reusable feature?

Looking forward to hearing comments!

Keridos, 2024-05-28:
Das Filament offers refills too. See #28. In there I simply set the non refill spools to have 250g of spool and the refills to have 0 grams of spool.

Roving-Ronin, 2025-01-26:
Did adding a "spool_type": "refill" eventuate / might it being added in future?