Donkie/SpoolmanDB #31: RFC: Glitter Filaments, how to define colors

Keridos, 2024-05-28:
Das Filament has some Glitter based filaments like this here: https://www.dasfilament.de/filament-spulen/pla-1-75-mm/405/pla-filament-1-75-mm-blue-pearl-800-g?c=33

It consists of black PLA filled with blue glitter. Technically it is multicolor but the current multicolor filaments needs an info how the color is spread. It is neither longitudinal, nor split in this case. 

How should these Types of filament be defined. We could add glitter or another modifier to the color spread info. Or we could defined filled filament types another way. But that should preserve the info about the two colors (base + glitter).

Donkie, 2024-05-29:
Do we really need to categorize based on glitter color? Or is it good enough to just specify a black filament with a "glitter" property.

Does this manufacturer produce filaments with different glitter colors but consistent base color?

Keridos, 2024-05-29:
As far as I can remember, Das Filament has different glitter colors, yes.

I think "Grüngold" is dark green filament with gold glitter.