Donkie/SpoolmanDB #32: Database Schema Refinements

YanceyA, 2024-05-29:
As people have started entering more filaments into the database there are have been some important findings and questions on how to enter some filament specific information. There is also a question on whether this information is important to be included in the data base to allow for categorization.

The goal is to have a bit of architectural discussion, discover additional areas to cover, and hopefully make PRs for a stable JSON schema. It seems that continuously litigating JSON schema changes is risky for database stability, uncertainty for data entry, inconsistent data entry.

As defined on the readme page "A centralized place to store information about 3D printing filaments and their manufacturers." seems to imply, to me,  that we are looking for a detailed database with lots of categories and details of filaments. However, this type of detailed database doesn't necessarily align with the core needs of a maker using Spoolman to catalog and maintain their physical spool library and support actual 3D printing. 

A non exhaustive list of open questions:

## Fills [see  https://github.com/Donkie/SpoolmanDB/pull/21 ]
- How should we handle filled filaments with carbon fibre and glass fibre. Other less popular and uncommon fills include wood, ceramics, kevlar, carbon/glass string, glow particles, metallic fill, glitter.
- Carbon and Glass fiber fills can also be specified in the % of fill and average fiber length.

## Glitter [see https://github.com/Donkie/SpoolmanDB/issues/31 ]
- Glitter is a technically a cosmetic fill.
- Glitter comes in different colors which provide various cosmetic affects.
- As far as i know there are no specifics for glitter size or type as it generally is not considered a benefit to physical properties.

## Refillable Spools [ https://github.com/Donkie/SpoolmanDB/issues/26 ]

## Material Names [ https://github.com/Donkie/SpoolmanDB/pull/15 , https://github.com/Donkie/SpoolmanDB/pull/9 ]
- Material names ideally should be presented at short names conforming to ISO 11469 and ISO 1043
- Material short names ABS are preferred over the long name "Acrylonitrile Butadiene Styrene" for obvious reasons.
- Can the material name be overloaded to catch

## Nylons or Polyamides [ https://github.com/Donkie/SpoolmanDB/pull/15 ]
- Nylons come in so many flavors (PA6, PA66, PA6.6, PA11, PA12, PA612....) 
- Specific standard ISO 16396 to deal with material marking of these materials
- Only a few nylons are actually used in 3d printing (currently) PA6 and PA12

## TPU:
- TPU come in different hardness across different scale ratings. This is a key parameter defining their function.
- Currently the guidance is to overload the material name to include the hardness rating "TPU-90A"

## References:
https://www.findoutaboutplastics.com/2020/12/plastic-part-marking-overview-codes-and.html
https://drive.google.com/file/d/1fIG9HlW-9AiyGDQsstiNUaXkKw-1Cyle/view

Keridos, 2024-05-30:
I'd add glitter to the fills section. Fills should be an array in case of multiple fills being present.

Fill object should be an object like:
```
fill-name:
  type: required
  color: optional
  size/length: optional
  fill_ratio: optional (because it can be unknown)
```
 
For TPU, I'd propose sticking to one standard and then standardizing it so it is "TPU-##A" to prevent confusion.
Regex that would be something like `TPU([0-9]{2}A){1}`
 
Special Filaments I would only broadly categorize those that with f.ex.
cleaning filament
mixed filaments (PC+ABS f.ex.)
flame retardant, though that could simply be an attribute for generic filament
support filaments, though that could be filament types, too
aero/lightweight

I think for the more esoteric filaments users should define them themselves, since you can also add custom attributes.

StuSerious, 2024-05-30:
https://regex101.com/r/SedDNb this is the current regex for matching material names, ad you can see it matches both shore A and D ratings ;)

Donkie, 2024-06-03:
I think my idea has been to support as much structured data as possible in the database, but only things that are necessary for Spoolman are required fields, such as density and spool weights etc.

At the same time, if we make the structured data too detailed then it just becomes too much of a chore to add. Fill for example could be enough to just specify fill type as we have now, and _maybe_ color. But on the other hand, fill color could be just as well specified in the filament color name itself. Specifying fill size and ratio is just way too detailed and I don't think anyone would search for exact quantities like this.

On another note:

I think having a GUI would make life easier for people to maintain this database, and I've been thinking of ways of doing that but haven't really come up with a good idea. I like the way it works now with people submitting PRs because it means we can utilize Github's built-in review functionality and don't need to create a new one. But then how to make a GUI that lets people edit the .json files and then submit as a PR? This is probably for a separate discussion though...

Keridos, 2024-06-03:
> At the same time, if we make the structured data too detailed then it just becomes too much of a chore to add. Fill for example could be enough to just specify fill type as we have now, and maybe [...]

Well, the ability to define what type of infill it is and assign that specific infill a color would be nice. I agree that we should not overengineer too much, though.

> I think having a GUI would make life easier for people to maintain this database, and I've been thinking of ways of doing that but haven't really come up with a good idea. I like the way it works [...]

Github offers codespaces so you can easily make changes in your browser with a web vs code session. Works pretty good for me, when I do not wan't to clone the repos to my computer.

Donkie, 2024-06-06:
> > I think having a GUI would make life easier for people to maintain this database, and I've been thinking of ways of doing that but haven't really come up with a good idea. I like the way it works now with people submitting PRs because it means we can utilize Github's built-in review functionality and don't need to create a new one. But then how to make a GUI that lets people edit the .json files and then submit as a PR? This is probably for a separate discussion though...
> 
> It would be amazing to add a tab to Spoolman's frontend where a user could either automatically submit their library or graphically input values to be converted in the JSON schema format.
> 
> Such an implementation is way above my head, as I have little experience with TS, but I could give it a shot. @Donkie if you're fine with a fair amount of hand-holding and reviewing I could try and submit a PR.

I'm a bit against having people sync their local spoolman library automatically to this database, since the quality of the data can be a bit hit and miss. It also would mean fields like fill type etc won't be added automatically since Spoolman doesn't have such fields built in.

runtpacket, 2024-12-05:
> I'm a bit against having people sync their local spoolman library automatically to this database, since the quality of the data can be a bit hit and miss. It also would mean fields like fill type etc won't be added automatically since Spoolman doesn't have such fields built in.

agree here unless we have some type of community moderation and polling when conflicting data

Truncated, 2025-02-17:
I've been thinking a lot about how I could or may want to use Spoolman in my overall workflow, and I'm finding a lot of potential integration and analytic points that could really allow amazing automations to be driven by the information it holds. Some things I'd like to see this project avoid:
1. **Allowing free text to take the place of figuring out complexity of schema and UI**. Basically relying on humans to put in data consistantly or correctly is a ticket to madness and sacrafices data integrity long term. The more value selections can be managed and relied upon to be consistant up front will pay massive dividends for everyone.
2. **Not considering UX impact of content scale resulting from Schema decisions.** What is easy to add / edit with five of something when they are all the same diameter may be very different when you have listings of every diameter and weight added as options.
3. **Not having a clear starting vision of how Spoolman is intended to be used.** I think it's done a good job of this to start, but as mentioned above there's some clear challenges in this growth and whatever its intended to target now should revise that vision.

So how do you use it now? 
What would you like it to do for you?

For my part, I came across Spoolman at the start due to it's clear lead with Inventory Management for FDM material. I've got two printers, I make my own models and print public ones, and it's definately a hobby for me. I am not at industrial scale, nor am I trying to make this a business or print farm.

I'm just trying to have something so I can do planning for projects, track my supplies, and help with the logistics around the lifecycle of deciding on a project through to getting it completed.

The current schema is pretty nice and simple enough to track filament, but I'm struggling with the flow of editing/updating and the lack of consistant treatment between the default schema and all custom fields in every way. The interface is a great start with a simple fundemental design, but with 50+ filaments I'm finding it a time sink and stumbling block more than assisting me in planning things and tracking anything without it being a whole new side-project.

So far, I'd like to know the following about my filament:

- Where and when I purchased it
- When I last dried it, and for each dry session to know for how long it was run, what method, weight before / after, humidity reading before and after.
- Who is the manufacturer
- Who is the supplier if rebranded (like MicroCenter does with their Inland brand)
- What is the material made up of?
- What kind of nozzle can I print it with? Size / hardness
- What kind of settings can I use (heat settings, flow, etc.)
- Whether it has common issues (shrinks badly, strings badly, etc.)
- Where is it stored?
- What is the color? (from a controlled source, like https://filamentcolors.xyz/library/)
- Where did I buy it from?
- what was the price of the spool?
- Is the spool empty?
- Is the product discontinued?

I think Spoolman currently can answer most of these questions, but not always efficiently. For me, most of these are to base logic from to filter by and integrate with printer control methods (OctoPrint add-ons, Orca Slicer, etc.).

Some of them have UX problems, others are hobled by how custom fields are handled and all of them might impact how the Schema is arranged. For instance:
1. Could current manufacturers be made to "nest" to reference the Supplier idea? Should that also include date rages to allow for changes in suppliers over time in 1:X methodology?
2. How many details about material can be made into check boxes, assuming the UX of the CRUD interactions can be smoothed out, instead of relying on a text field to hold all of those differences?

**So what are your use cases? What are the things you're trying to do or finding issues with as it stands?**

Mind you, I am perfectly okay for this project to say "well, what you're about isn't the goal". I'm not trying to insinuate that any of these needs are even expected, let alone required. I'm just giving my own context and thoughts to start the ball rolling.

kquinsland, 2026-01-19:
> So far, I'd like to know the following about my filament:
>
>     * Where and when I purchased it
>
>     * When I last dried it, and for each dry session to know for how long it was run, what method, [...]

I am about 95% overlap with those questions specifically as I look for things that can integrate with [OPT](https://openprinttag.org/).

It might be worth taking a look at how [Inventree](https://docs.inventree.org/en/1.1.x/api/schema/) (used for stock keeping for electronic parts) does this.

If you squint a little bit, a spool with filament can be modeled a _lot_ like electronic components. Just like "some ABS" has a ton of variance, so does "a resistor".

Likewise, _where did `$thisComponent` come from_ usually has a non-trivial custody chain... like with filament that might be manufactured by an OEM and re-branded and then bundled into a new SKU.