Donkie/SpoolmanDB #33: Feature request - Add Safety Data Sheet & Technical Data Sheet

TexZeTech, 2024-06-03:
I would like the integration of Safety Data Sheet & Technical Data Sheet information with links to original source.

StuSerious, 2024-06-05:
I like the idea, but I would suggest SDS and TDS files be placed in a folder in this repo for archival purposes instead of using hyperlinks subject to change by the manufacturer.

Donkie, 2024-06-06:
Yes as Stu said, we should probably host these ourselves to prevent links going stale. I haven't figured out a nice system for this yet though.

Nojwe, 2025-05-14:
I know this is an old one, but I was thinking about something similar the other day. I think we’d need to use a web scraper to pull SDS and TDS from the manufacturer sites. Ideally (to me), the files would be archived, parsed, and the relevant data stored in SpoolmanDB templates (density, diameter, recommended temps, etc.). Those of us using tools like spoolman2slicer can use that data to automatically create filament profiles too.