Donkie/SpoolmanDB #57: Pulling info into Spoolman

limited660, 2024-12-04:
Is there anything covering how to get these into the Spoolman instance? I've searched the spoolman docker install and cannot find where the json files are kept.

ali86x, 2024-12-07:
Would be interested true. 
I did Not find a way to add the Data to spoolman.

zacgoodwin, 2024-12-10:
When you add the filament or spool it has both an internal and external section. The external section pulls from this Github Repo

[image]

[image]

ali86x, 2024-12-10:
Sometimes it could be so easy 😂

Thank you @zacgoodwin

limited660, 2024-12-10:
Yes, very simple now that I see the button! Thank you.

closed 2024-12-10 by limited660