Donkie/SpoolmanDB #76: Adding Filament

nightbrightly, 2024-12-31:
Would it be possible to add Ambrosia Filaments that west3d makes they are really nice. I'm not that savvy on github but would love their stuff to be added.

https://ambrosiafilament.com/

StevensStuff, 2025-01-08:
I'm currently working with West3D to get all the hex values.  As soon as they are received then ambrosia will be PR'd

RoyalNautilus, 2025-06-09:
@StevensStuff did you ever make any progress with this? Just got my first few rolls of Ambrosia in and would love to add them to Spoolman. I'm happy to do the work if you have access to the hex values.

StevensStuff, 2025-06-09:
> [@StevensStuff](https://github.com/StevensStuff) did you ever make any progress with this? Just got my first few rolls of Ambrosia in and would love to add them to Spoolman. I'm happy to do the work if you have access to the hex values.

I have about 1/2 the hex values, I've been waiting on West3D with no luck.  I'll try to guess and get at least close values for the rest this week.

Donkie, 2026-10-26:
Many have been added now. Please add some yourself if anything is missing!

closed 2026-10-26 by Donkie